Reisyukaku/ReiNX #5: Missing source code for kips in data/?

kitlith, 2018-07-22:
sm.kip and loader.kip are included in the repository, but appear to have no source code to go with it.

Either these are builds of your own modules, in which case action is not required (though I think people would like to see the source code), or they are builds of Atmosphere's modules, licenced under GPLv2, and at the very least the commit used to build these should be noted somewhere. If they have been modified, please post the source code, etc.

M1CR0H4CK3R, 2018-07-22:
From README.MD's [initial commit](https://github.com/Reisyukaku/ReiNX/blob/9625fd163f0aa9f285d69b353988d9b7b0348388/README.md):

> Credits:
> 
> Naehrwert for hardware init code and generally being helpful!
> 
> CTCaer and st4rk for their contribution to the hardware code aswell!
> 
> --**SciresM for kernel processes!**--
> 
> The community for your support!

So I think that we can assume that he **did** rip these files from Atmosphere. @Reisyukaku, @kitlith is right, you should make it much more clear that those files were not your own. You credited SciresM, but did not state whether you used the files in whole, in part, or if the files are just using the same naming scheme as his "kernel processes". Either way, you need to be more clear. 

This is especially true when you are developing software of this nature; for example, a group of "hardcore TX fanboys" (believe me, those can and do exist) could easily get this repo taken down by reporting it for GPL violation (even if just for a few hours), and (although this could happen regardless of whether or not you violate GPL, but this snippet is about being careful with regards to the nature of this software, and still applies here) the company who developed the hardware that this code is designed to run on could, if they really wanted to, send a C&D against your project to you and have your GitHub accound closed if you don't remove the code in its entirety (although this is unlikely, knowing the above indirectly mentioned company, they could if they really wanted to, and we don't want them to have "as good a time as any" to do it.

TL;DR - @Reisyukaku, you're already treading dangerous waters; please be more precise with your credits regarding sm.kip and loader.kip in your readme file.

Reisyukaku, 2018-07-23:
https://github.com/Reisyukaku/NX_Sysmodules
ill link them in the future

closed 2018-07-23 by Reisyukaku

kitlith, 2018-07-23:
Hey, sorry @Reisyukaku, I didn't see the notice about not done uploading stuff. Apparently I didn't read enough. Someone else pointed it out, and I wanted to make sure there was an actual issue filed instead of accusations of GPL non-compliance flying around. Was not my intent to make a big deal out of anything. :sweat_smile: