Reisyukaku/ReiNX #11: bad return value

SegFault42, 2018-07-23:
Hello,

Why you return NULL in your fopen function instead of -1 or 0 ?

The function return a u32 value. 

I have several warning when i compile the code.

e.g :
```
src/fs.c: In function 'fopen':
src/fs.c:45:16: warning: returning 'void *' from a function with return type 'u32' {aka 'long unsigned int'} makes integer from pointer without a cast [-Wint-conversion]
         return NULL;
                ^~~~
```

https://github.com/Reisyukaku/ReiNX/blob/2bee04accf4fed5e0e96be8d0778fc923f4e7618/src/fs.c#L43

Reisyukaku, 2018-07-23:
In C, Null is defined as 0 in stddef.h The warnings are for something different. I'll clean it up eventually.

SegFault42, 2018-07-23:
A macro that expands to a null pointer constant. It may be defined as ((void*)0), 0 or 0L depending on the compiler and the language.

frome here : https://en.wikibooks.org/wiki/C_Programming/stddef.h

Reisyukaku, 2018-07-23:
right, ill just have the function type be void pointer than.. should fix it

Reisyukaku, 2018-07-24:
should be fine

closed 2018-07-24 by Reisyukaku