Reisyukaku/ReiNX #18: SDSC cards not supported

TheDgtl, 2018-07-27:
Tested by multiple users in Discord, ReiNX doesn't seem to support SD cards smaller than 4GB (Tested with 2GB and 64MB, both return "Failed to init SD card")

Most likely an SD vs SDHC/SDXC but really, who would run a 2GB or smaller SD card in a Switch? :)

Note: Same issue occurs in Hekate. Returns a FAT error 13

Reisyukaku, 2018-07-29:
That will be difficult to test in the current year ;^)
Does it work on hekate?

TheDgtl, 2018-07-29:
Hah, I know, right? Surprisingly me and another user both had the same 64MB card to test with.

Fails in both Hekate and ReiNX, so I wouldn't be against just calling it a known issue. Only opened the issue for tracking purposes, to point to if anyone else complains :)

Reisyukaku, 2018-08-03:
I'm gonna close this for now because Switch normally doesnt support SDSC anyways, but if someone sends a PR or implements it in hwinit, i'll reopen.

closed 2018-08-03 by Reisyukaku