Reisyukaku/ReiNX #28: XCI backup loader

lfpereira, 2018-08-04:
@Reisyukaku, please make a .XCI loader for us.
It's possible?
Thanks!

lolicop123, 2018-08-05:
it would be nice if this was incorporated into the ReiNX Homebrew launcher

lunalik2, 2018-08-06:
He is already working on it

lfpereira, 2018-08-07:
For me, one of the main advantages of XCI is being able to remove the microSD, and play the original games online, without displaying pirated NSP icons on Home. I think it's much easier for Nintendo to identify fake NSPs installed than save games from XCI-loaded games. In addition to the ease of use, you simply paste the files into the microSD without losing installation time and double space as you need before removing the NSP after installation.
That's why we need XCI loader...

HugoOdaX, 2018-08-07:
@lfpereira Yeah, we need it but we have to wait a little bit

davisrayler, 2018-09-02:
Please XCI Loader :)

psfree, 2018-09-05:
We are working on something.

lunalik2, 2018-09-05:
@psfree I hope all goes well :) This should shove the stake in SX’s grave.

doomtrigger, 2018-09-26:
june 15th

lunalik2, 2018-09-26:
@doomtrigger I have a feeling you created a Github account just to say that.

doomtrigger, 2018-09-27:
I did lmao

maciver, 2018-11-30:
lol

azoreseuropa, 2018-12-20:
I hope that you are still working on it. Keep up the good work, ReiNX team. I love ReiNX!

Multimegamander, 2018-12-23:
Atmo is SX OS clone confirmed

maciver, 2018-12-23:
> Atmo is SX OS clone confirmed

Are you stupid?

azoreseuropa, 2018-12-23:
Take it easy. I didn't know that Atmo is SX OS clone confirmed. I just
found out about it at gbatemp. Chill out, man. SMH.

azoreseuropa, 2018-12-23:
Yeah, good to know. I just found out about it at GBATEMP. Thanks.

Multimegamander, 2018-12-27:
@Reisyukaku why delete his comment, the Nazis tried to keep people quiet, I see straight through you. Nazi!

maciver, 2018-12-27:
> @Reisyukaku why delete his comment, the Nazis tried to keep people quiet, I see straight through you. Nazi!

So you are only here to troll? GTFO then, some people actually want to be here to keep up with real issues.

Multimegamander, 2018-12-29:
Jesus is always knocking on your soul, it's your choice if you want to let him in, and show you the light, or not.

azoreseuropa, 2018-12-29:
Enough with drama. Be nice to each others, alright ?

Multimegamander, 2019-01-05:
> Enough with drama. Be nice to each others, alright ?

no

lavaJockey, 2019-06-16:
Can someone just close this tbh. Just use nsps.

lfpereira, 2019-06-17:
> 
> 
> Can someone just close this tbh. Just use nsps.

Friend, if for you the resource does not seem useful, not to say that it is not useful to others.
My 128GB microSD is not enough for all games, so I use external HD with several XCIs that are loaded by SXOS.
In this case an XCI loader is very useful since NSPs can not be installed on the external hard drive.

xomniversex, 2019-06-21:
As someone that prefers XCI loading I bought my first switch and SX OS Pro kit but it did indeed leave a bitter taste in my mouth for financially supporting DRMed proprietary software so I have faith in XCI loading in open source implementations. With Atmos EmuMMC at least EmuNAND is no longer unique to SX. That leaves just XCI. So thumbs up for any effort towards that end. I hope to one day kick SX OS to the curb.