Reisyukaku/ReiNX #38: Game Crash Once 32 Launches

addictedgmr, 2018-08-16:
Me and a few people have reported an issue with the sigpatch we use. We need more people to test. But the issue is - over time in a boot of any CFW of your choice. Launching a game (the same game or different games) 18 times (sometimes more depending on the CFW) causes a crash. 
Sleep mode does not affect it. It seems to just be closing a game and launching another. 
The counter does not reset with sleep mode. Only with a reboot. 
My stats have show SX as 18, reiNX as 32, Atmosphere as 32 and **Exosphere as 35**. It may vary though.
On SX it seems to say software closed, and then any game you launch, error to 2144-0001 ```If you have time, boot into a CFW (SX, Rei or Hekate), and open and close a game 20 times.```

Melon-Bread, 2018-08-17:
I can also confirm these numbers

Reisyukaku, 2018-08-26:
has anyone tried without fs_mitm?

angelmarlboro, 2018-08-26:
What is that file for? What would happen if I erase it? just for learning what I'm doing

addictedgmr, 2018-08-26:
> Hello Addictedgmr thanks for your ansewer. i don´t have my nintendo right
> here to test this tip. can you explain me what it is LayeredFS? and what
> happen if i delete this file? thanks again

it allows for game modification. We can overlay game files with modified
ones. It’s only if you set it up correctly. Like you could overlay Mario
jump distance value for example.

angelmarlboro, 2018-08-26:
OK, i understand, so this meens is use to modify propperties of the game, like jump distance or damage? i dont use layeredfs. so... can I erase it?

addictedgmr, 2018-08-26:
Yes, easily. And it’s incredibly easy to erase. The mods are placed on your
SD and when you open the game with that layered kip and the mod files set.
It works fine.

angelmarlboro, 2018-08-27:
ok, in my case the file fs_mitm is not in my sd, and same situation after 20 lounches aprox. mi console freeze, with no error at all, just freeze. i push power button 15 seconds and then i inject rainx.bin payload.
change mi sd memory, to another, i formated with nintendo cfw , same error. thanks for you help and time again

Reisyukaku, 2018-08-29:
ill have to make better FS patches i suppose.

kanielyaniel, 2018-11-04:
This happened to me, but instead of a full crash, it just makes the game unrunnable (error every time) and when I open the hbmenu, it gives me an error. After, the ReiNX splash gets corrupted.

Reisyukaku, 2018-11-10:
thought i already posted here but its fixed in latest

closed 2018-11-10 by Reisyukaku