Reisyukaku/ReiNX #51: Issues with booting on 4.1.x

neonsea, 2018-09-08:
There are reports of people being unable to boot on firmware 4.1.x.
https://gbatemp.net/threads/official-reinx-thread.512203/page-76#post-8268192
https://gbatemp.net/threads/official-reinx-thread.512203/page-76#post-8268224

Notably, these people have manually installed exFAT support.

Dstr8ction, 2018-09-08:
I didn't touch my 3.0.0 Switch for about a year hoping for a non-tethered coldboot exploit.

A few weeks ago I lost my patience and finally wanted to use my Switch. That was when I read that at least the non-released warmboot exploit Deja Vu will support consoles up to 4.1.0 and thats why I chose not to update to 5.1.0.

shchmue, 2018-09-08:
I don't understand enough about Switch architecture or the codebase to speculate why this would be but I'm wondering if since that patch only rewrites the pkg2 the loader might be doing something wrong. I'm curious if re-doing the exFAT patch with ChoiDujourNX alleviates the problem.

gihucom, 2018-09-08:
Hi,

i am gt10 from gbatemp (the other guy with the same problem)

i had black screen before with newer versions (reinx) included in darth meteos stash.
like i described in this thread https://gbatemp.net/threads/black-screen-after-reinx-splash.515863/
summary of the thread:

- my newly buyed nintendo switch came with firmware 4.1.0 installed (without exfat support)
- with darth meteos stash v2.2 reinx worked (using exfat sd card), but it contains fs.kip that seems to be not very legal.
 - with darth meteos stash v2.3 the black screen problem began.
- but v2.3 with fs.kip keeped on the sd card made reinx start.
- another user wrote it could be a problem with exfat support.
- to get rid of fs.kip i installed exfat support on 4.1.0 with homebrew ChoiDujourNX v1.01
- now v2.3 of meteos stash started without fs.kip

after that i kept v2.3 until now. 
with seperation of darth meteos from ReiSwitched and the new release of reinx 1.6 with new features/tools i wanted to update the cfw.
but now i again have the issue with only booting to a black screen.
to see if its a reinx only problem i tried darth meteos stash v5.0 with also booting to black screen.

thank you for trying to help with my/this problem.

Endar-Marit, 2018-09-08:
@shchmue didn't fix for a user in the discord. getting them to post here

FranTatachan, 2018-09-08:
I updated my switch from 4.1.0 without exFAT to 4.1.0 with exFAT.

Before that, I had to put the FS.kip file inside sysmodules folder when I ran ReiNX 1.2. Now it works without the FS.kip file.

When I try to run ReiNX 1.6, the splashscreens appers, but after that the screen keeps black.

I don't use autoRCM

neonsea, 2018-09-08:
Further reports seem to suggest that exFAT isn't related to this issue, as people on 4.1.0 can't boot even without having installed exFAT support.

Reisyukaku, 2018-09-08:
Seems the consensus im getting is the kips are breaking things. Which is very strange.

Reisyukaku, 2018-09-08:
Ok i think i found the issue and it should at least boot now as of 2e2fab91255fcb5087ee92035d4b7069fea6d787

Ill find a better pattern soon.

closed 2018-09-08 by Reisyukaku