Reisyukaku/ReiNX #57: Silent launch

HowManyNamesHaveBeenTaken, 2018-09-11:
Not actually an issue, but having a way to launch Reinx without seeing the small text at bootup would be nice.  Either a launch option or an alternative RCM payload if its coming from the bin. Thanks for the CFW!

neonsea, 2018-09-11:
This is already the default behaviour since https://github.com/Reisyukaku/ReiNX/commit/2cc51b5cfe3d85fbbee3b79ba9f5cf4bc660f40c

Only booting while holding VOL- should display text. Update your ReiNX.

HowManyNamesHaveBeenTaken, 2018-09-11:
Am using the latest build as of Sept. 11th, just started using ReiNX today. Still get text at bootup.

neonsea, 2018-09-11:
That means that the ReiNX.bin on your SD is still an old build. Delete that file or update using the toolkit.

HowManyNamesHaveBeenTaken, 2018-09-11:
There was no ReiNX.bin on my SD card, thought it was just a RCM payload. Just put the latest one on there and still get text.

neonsea, 2018-09-11:
🤔

This shouldn't happen at all. What happens if you boot ReiNX while holding volume down, exactly the same behaviour? Or something else?

HowManyNamesHaveBeenTaken, 2018-09-11:
Oh, that is a weird way to handle it, alright. Thanks for the help.

closed 2018-09-11 by HowManyNamesHaveBeenTaken