Reisyukaku/ReiNX #65: Modified NSPs appearing as "corrupted" when launched

PsychOsmosis, 2018-09-24:
Hello,

I released a batch script a week ago to modify NSP files so the title rights are removed, the header is rewritten to no longer read the ticket file and the ticket is deleted.

It works flawlessly on SX OS (the CFW I personally use), but it's been reported to me that people on ReiNX can't get the modified NSPs to run on their console. When they launch the title, they get a message that the game is corrupted and are prompted to connect to Big-N's servers to redownload the game.

I searched online to understand why that's happening or if there is a fix that was released for this issue, and I've read that since about a month ago, ReiNX supports modified NSPs.

The script uses nut, which is itself a script that uses hactool to modify the NSPs.
The issue occur both on firmware 5.1 and 6.0 (two different people on two different consoles).
Also, the issue occurs even on the latest ReiNX nightly build.

Do you know why this happens and/or do you know what these people can do to fix it?
I don't know much about ReiNX since I personally don't use it, but is there some modified sysmodule these people should install in order for modified NSPs to be supported?

Thank you very much in advance for your help in this matter!!

Reisyukaku, 2018-09-26:
ReiNX works with all nsps ive thrown at it.. Only thing I can think of is its a 6.0 thing because thats literally the anti-piracy update lol.

PsychOsmosis, 2018-09-26:
Hmm... I don't doubt it, but would you mind applying the script to a NSP and trying it out on ReiNX? If it indeed works, it means those people both have had the same outside problem. I can understand the person on 6.0 having issues (I also thought it was the culprit), but the other person is on 5.1.

Here a link to the page: https://gbatemp.net/threads/how-to-remove-fake-tickets-from-nsp-file-standard-crypto-equivalent-ban-risk-mitigation.518312/

The prerequisites are quite simple, download nut from github, have python installed with a few modules (which you probably already have), and pasting keys.txt and the .nsp in the nut folder. Then you create your .bat (also in the nut folder) by copy-pasting the script code and finally you drag&drop the .nsp onto the newly created batch file.

I understand if you don't have the time to try it out though, but it would really help me in figuring out why those people can't get it to work, when the script seems to do its work just fine.

I really appreciate that you took the time to answer my question! Thanks you very much!!

closed 2019-04-28 by Reisyukaku