Reisyukaku/ReiNX #66: fs_mitm and ftp issue won't start

vaccarieli, 2018-09-26:
When I try to use them the console wont start, K have to either use fs_mitm or ftp, one sysmodule at a time.

Reisyukaku, 2018-09-26:
I've never used ftp but I dont see why they would interfere with eachother.

vaccarieli, 2018-09-26:
Well actually I use it due to my son, because he likes being going around the whole menu and goes to homebrew so I got scared one time, so the easier way for me to deactivate homebrew is by going to the title folder and rename it, so it can not access and then when I need it just do the same :) lol

F3Speech, 2018-09-30:
@vaccarieli I've also found sys-ftpd.kip not booting with mitm but you can swap it out for ftpd from the hb app store and that works fine.

v0dKa-mL, 2018-10-15:
I have the same Problem both together the Switch won´t boot

Reisyukaku, 2018-11-18:
could try latest bcfb58dcba87d49103725216f67572d2c8c2bed8

closed 2018-11-26 by Reisyukaku