Reisyukaku/ReiNX #72: Idea: Reinx's own fw upgrade function

khnoize, 2018-09-30:
So, since just booting cfw itself would get the superban flag. The only way to update fw for cfw users are only a certain homebrew. And since this certain homebrew developer seems to not having a good relationship with reinx, I'm just thinking that having this functionality at reinx itself would be very great addition.

Reisyukaku, 2018-09-30:
I dont think I follow.. I just update via system settings lol. Never had to use ChodeDigorno or whatever.

khnoize, 2018-10-03:
Um, okay. I'm actually haven't tried yet, but I heard that even just booting cfw right now would means superban, so, no fw upgrade. But since you still use it, then that's not true right?

Reisyukaku, 2018-10-07:
You're gonna get banned tampering with the system in any way.. so updating via nca installs or whatever isnt gonna save you. As soon as you hit online, you'll be flagged. Being constantly offline is basically no different than just being banned lol. Either way, Raj being salty shouldnt effect its usage with rnx though.

khnoize, 2018-10-08:
@Reisyukaku I know, so basically, we ReiNX users are "already banned" (since (offline/banned) == no official nintendo services), and no more official system fw updates in the future, right? I was just wondering if it's great idea if ReiNX have its own fw updates tool since Raj being salty. Either way it's ok, we probably just would be have to use the existing tool and continue using ReiNX. Anyway, thank you for your responses.

Reisyukaku, 2018-10-10:
There are ways to update even if banned but i have to see if i can streamline that

khnoize, 2018-10-15:
Thanks, It would be great if this does come true one day. Though probably there are more immediate priorities than this for Reinx, hence it's just an idea.

Multimegamander, 2019-01-05:
@Reisyukaku how long do I cook a ChodeDigorno for, senpai?

closed 2019-03-19 by Reisyukaku