Reisyukaku/ReiNX #79: 6.1.0 (possibly) breaks AutoRCM

spoonh, 2018-10-30:
Updated to 6.1.0 while using AutoRCM, my switch does not boot into AutoRCM anymore and disabling/re-enabling auto rcm does not seem to do anything. No idea if i'm the only one with this occurrence.

Faerye0, 2018-10-30:
Same problem here

noahc3, 2018-10-31:
OP said disabling/re-enabling doesn't work.

Reisyukaku, 2018-11-01:
Because they're probably enabling and disabling again.. just do it once. All autorcm does is write to the RSA mod on the BCT in nand. It literally bricks your nand forcing the system into RCM mode. lol

spoonh, 2018-11-01:
> Because they're probably enabling and disabling again.. just do it once. All autorcm does is write to the RSA mod on the BCT in nand. It literally bricks your nand forcing the system into RCM mode. lol

This works, wasn't paying enough attention when i did this the first time i guess.
Thanks for your help.

neonsea, 2018-11-01:
Great stuff

spoonh, 2018-11-04:
Having this issue again, switch is essentially in "bricked state" after disable/reenable but is recognized as Nintendo Switch rather than 'APX'. Any ideas?

Reisyukaku, 2018-11-04:
uhh try fully powering it off (holding power for 20 seconds) and turn it on?

spoonh, 2018-11-04:
> uhh try fully powering it off (holding power for 20 seconds) and turn it on?

that works, but i have to do this every time i turn the console off now. eh, i'll just deal with it i guess.

closed 2018-11-04 by spoonh