Reisyukaku/ReiNX #83: There are some compilation errors with the lastest libstratosphere

JamePeng, 2018-11-10:
it looks like the new libstratosphere had refactored the file structure, the NX_Sysmodules file still used the old version include to implement the function.

/ReiNX/NX_Sysmodules/loader/source/ldr_debug_monitor.hpp:4:10: fatal error: stratosphere/iserviceobject.hpp: No such file or directory
 #include <stratosphere/iserviceobject.hpp>

Reisyukaku, 2018-11-10:
Yea i thought i'd be nice and just link to that but seems to be causing more harm then good.. so fuckem. Reverted.
also using Boost is a good sign of a filthy noob ;v)

JamePeng, 2018-11-10:
lol, but  this fix still need the scope_guard.hpp of the boost~

ON_SCOPE_EXIT {
        /* Loader doesn't persist the copied resource limit handle. */
        svcCloseHandle(reslimit_h.handle);
    };

Reisyukaku, 2018-11-11:
Ah right, I wasnt in the mood to deal with these memes yesterday.. looks like their submodule has submodules and for whatever reason just using git clone recursive submodules or whatever doesnt work alone so i went under NX_Submodules and ran
`git submodule update --init --recursive`
because they  have custom boost shit under include
Anyways i reverted it
73c66b73c4193e23d4da6e9d795f3b42248267b6
just pull that and make sure that /include/boost/ dir has shit in it

Reisyukaku, 2018-11-11:
theres still a few loose ends i gotta figure out when i get home it seems.. will reopen till then

Reisyukaku, 2018-11-11:
So uhhh things compile but dont boot because a lot of the updated code is atmo specific shit so idk when ill get around to figuring that out.

closed 2018-11-11 by Reisyukaku

JamePeng, 2018-11-12:
yes, use the lastest compile will boot failed! and I have to open the switch shell remove the battery for release the power.

Reisyukaku, 2018-11-18:
Everything should work as of  bcfb58dcba87d49103725216f67572d2c8c2bed8

Reisyukaku, 2018-11-18:
also you dont need to open the switch lol.. just hold power button for 20 seconds