Reisyukaku/ReiNX #85: compile faild

Ts1ng, 2018-11-12:
aarch64-none-elf-g++ -MMD -MP -MF /home/centos/opensource/ReiNX/NX_Sysmodules/loader/build/ldr_process_manager.d -g -Wall -O2 -ffunction-sections -march=armv8-a -mtune=cortex-a57 -mtp=soft -fPIE -DDISABLE_IPC -I/home/centos/opensource/ReiNX/NX_Sysmodules/loader/include -I/opt/devkitpro/portlibs/switch/include -I/opt/devkitpro/libnx/include -I/home/centos/opensource/ReiNX/NX_Sysmodules/loader/../libstratosphere/include -I/home/centos/opensource/ReiNX/NX_Sysmodules/loader/build -D__SWITCH__ -fno-rtti -fno-exceptions -std=gnu++17 -c /home/centos/opensource/ReiNX/NX_Sysmodules/loader/source/ldr_process_manager.cpp -o ldr_process_manager.o 
/home/centos/opensource/ReiNX/NX_Sysmodules/loader/source/ldr_process_manager.cpp: In member function 'std::tuple<unsigned int, MovedHandle> ProcessManagerService::create_process(u64, u64, CopiedHandle)':
/home/centos/opensource/ReiNX/NX_Sysmodules/loader/source/ldr_process_manager.cpp:39:5: error: 'ON_SCOPE_EXIT' was not declared in this scope
     ON_SCOPE_EXIT {
     ^~~~~~~~~~~~~

JamePeng, 2018-11-12:
Two temporary measures:

First, update the submodule Reisyukaku/NX_Sysmodules to 8db3a8b.

Second, modify the Makefile of reinx root.like this:
line 58    @cp $(dir_sysmod)/pm/pm.kip $(dir_out)/ReiNX/sysmodules/
line 59   @cp $(dir_sysmod)/fs_mitm/fs_mitm.kip $(dir_out)/ReiNX/sysmodules/

Reisyukaku, 2018-11-18:
this is already being discussed https://github.com/Reisyukaku/ReiNX/issues/84

closed 2018-11-18 by Reisyukaku