Reisyukaku/ReiNX #86: Compression for NAND dumping

Warfields, 2018-11-14:
The longest part of the NAND dump is the disk transfer, Since all of the dump files are ~3 Gb is it possible to load it into the Switch's ram and then run compression on it before saving to the SD? Just by using a zip compression i was able to reduce the 30Gb dump down to 604Mb after it was transfered to my machine, this would greatly increase the speed of the NAND dump.

haxxey, 2018-11-14:
NAND dumping is included in the ReiNX Toolkit, not ReiNX itself - you should post the issue on its issues page https://github.com/Reisyukaku/ReiNXToolkit/issues

neonsea, 2018-11-14:
NAND dumping is included in the ReiNX Toolkit, not ReiNX itself - you should post the issue on its issues page https://github.com/Reisyukaku/ReiNXToolkit/issues

Warfields, 2018-11-15:
> NAND dumping is included in the ReiNX Toolkit, not ReiNX itself - you should post the issue on its issues page https://github.com/Reisyukaku/ReiNXToolkit/issues

Thanks I will do!

closed 2018-11-15 by Warfields