Reisyukaku/ReiNX #88: nintendo switch 6.2 update

GamingKingKKu, 2018-11-20:
all CFWs are broken for this version.

crdnl, 2018-11-27:
Small update: 6.2.0 updated key generation

new keys posted https://twitter.com/elmirorac/status/1066106288747237377

Jacoby6000, 2018-11-27:
Those are key hashes, not the actual keys.  Those are being kept under wraps at the moment.

ALANARIAS, 2018-11-29:
Does anyone know when the new reinx update will come out?

Will it be a long or short time?

Reisyukaku, 2018-11-29:
Not long

QuackerJ4ck, 2018-11-29:
Rei do you know how long exactly? I'm sorry I just really wanna play the new smash I really appreciate all the work you've done for us

crdnl, 2018-11-29:
@QuackerJ4ck Then go preorder the game. It will run on FW 5.1.0 and up without issue, you just can't play online.

QuackerJ4ck, 2018-11-29:
Wait how does that work would I have to downgrade my firmware?

Jacoby6000, 2018-11-29:
Nobody knows how long. If you want cfw on 6.2 you need to wait. If you’re currently on 6.2 and you want CFW now, you need to downgrade.

Reisyukaku, 2018-12-01:
We're shooting for this weekend.

ALANARIAS, 2018-12-02:
This weekend, will the update come?

jamescastells, 2018-12-02:
Hey, guys! Since atmosphere was updated, is that good news for Reinx? By the way, fantastic work!

balika011, 2018-12-03:
Weekend is over.
Where is the update?

bell07, 2018-12-03:
ReiNX is open source, if Rei is not fast enough for you, get the source, hack the 6.2 and release the awaiting update. Do not forget to set up a pull request.  ;-)

piratesephiroth, 2018-12-03:
I guess they missed the shot

tmargemZ, 2018-12-03:
I Just Hope It realease soon, bought my switch yesterday played 30 mins of Kirby then accidentaly update to 6.2... shit happens

Nekxes, 2018-12-03:
> I Just Hope It realease soon, bought my switch yesterday played 30 mins of Kirby then accidentaly update to 6.2... shit happens

You still can use atmosphere to play any NSP games/app !

VersatileNinja, 2018-12-04:
It's done when its done. No pressure. Take your time team!

HappyVideoGameNerd, 2018-12-04:
Seriously...
It's fine to be excited, and even to wonder about progress.  But the man behind this doesn't owe us an exactly release date.  He doesn't even owe us the firmware at all.  He's doing it to be a bro.  

So chill out, and ease up on the "are we there yet"s.  Less gimme-gimme and more gratitude.  

It'll be done when it's done.  I'm excited too, but spamming the board with "release plz" only makes people mad.  And considering this is a project for little to know personal gain for the devs, we should really avoid being ungrateful.  

So to everyone involved in 6.2 and the work that went before - thanks.  We appreciate it.

Romeosote, 2018-12-04:
Gracias reí confiamos en ti

Chara-Reborn, 2018-12-09:
Shouldn't this be marked as closed? 🤔

closed 2018-12-09 by Reisyukaku