Reisyukaku/ReiNX #91: Failed To Init SD Card

myfire888, 2018-12-16:
So, I have 3 microsd that worked perfectly before.
today I was just playing mario rabbids, and then I want to play on the tv. so I put it on dock, and when I went to open the game again, it hangs.
so I turn it off, and then I change the microsd (because I decided to try other games on other microsd). but then when I inject the payload (with r4i dongle), it blank for a while, and then error with red notes "failed to init sd card"
now all 3 microsd couldnt be used.

but I can read the microsd on pc, and switch ofw could read them too (obviously not trying to play the nsp but I can see how much storage used and free, and like that)
I suspected maybe the version got updated because I go online before to try apps inhomeswitching (but I have made sure to fill the dns to block ninty server).
but no, the version still 5.1.0 like before. 

so now I'm in the dark. dont know what to do. I have tried to update to the latest reinx that supported 6.2.0 and still the same error.

pls help

Reisyukaku, 2018-12-17:
Are you sure your SD is reading on OFW? because sounds like a hardware issue. Especially if it just stopped working.

myfire888, 2018-12-17:
well, on ofw, if I open setting - storage, I can see the remaining size, and all the games that I have installed with the size on them (of course I cant open any of the games cause its on ofw).  I think that indicates that the sd was read on ofw ?

myfire888, 2018-12-17:
because I used 3 microsd, so from what I know, on the setting - storage/games, the games listed would have 0.0 kb on their size if the game is on the other microsd. and the game with the size on them is what is on the microsd that's put on the switch

tbennett6421, 2019-02-26:
Hi @Reisyukaku 

I'm experiencing this as well. I have a known good SD card and purchased a new one to get more storage. Are you able to verify that 256GB sd cards work in ReiNX? The SD card is supported in official firmware and was initialized in OFW.

Known Good
* PNY
* Elite microsdxc
* Class 10, U1
* 64GB
* https://www.pny.com/Elite_microSDXC_Card_with_OTG_Reader

Experiencing Issues
* SanDisk
* Ultra microsdxc
* Class 10, U1
* 256GB
* https://www.sandisk.com/home/memory-cards/microsd-cards/ultra-microsd-400gb

I wanted to check before trying to return/RMA the card. It may be worth adding to a known bad sd card section if this is unreliable for reinx

Reisyukaku, 2019-02-28:
Our SD drivers are based off nintendo's. I personally use 128GB and that works fine so long as its formatted properly.. i have it formatted as FAT32. Are you sure OFW is actually reading it properly? just seems bizzare. Did you see if hekate can read the SD?

mramirkhan, 2019-03-19:
I've been using Samsung 256gb evo microsd cards with this cfw on multiple switches without issues.

tbennett6421, 2019-03-19:
I spoke with SanDisk on the issue and the card I was using was a knockoff using the SanDisk brand name.

Reisyukaku, 2019-03-20:
Ah, yea. Watch out for chinese knockoffs.
I'll close for now since im sure that was probably the issue

closed 2019-03-20 by Reisyukaku