Reisyukaku/ReiNX #97: New Super Mario Bros U Deluxe "Corrupted data"

PaS-NaS, 2019-01-11:
New super Mario Brow U Deluxe was recently launched and it doesn't work on my switch with Reinx 2.0  (firmware 5.1.0). As this game works when I launch atmosphere I guess this could be easy to solve. Whenever I try to launch it with Reinx I get the message of corrupted or missing data.

Reisyukaku, 2019-01-26:
Considering atmo uses my patches, im sure its not a ReiNX problem

PaS-NaS, 2019-02-21:
Well, I updated (offline) to 6.1 and now Reinx makes an error when I launch it and it makes my swtitch reboot. Atmosphere still working. I really prefer Reinx and I don't knwo what to do. Obviously I used the newest files from your site but still, not working

Reisyukaku, 2019-02-24:
Make sure you delete reinx directory and follow the guide if need be. https://reinx.guide
everything should work out of the box.

PaS-NaS, 2019-02-25:
I tried to delete de reinx directory instead of overwrite and now it works perfectly. Now I can play New Super Mario Bros U Deluxe with Reinx. Maybe the game wasn't compatible with 5.1 Switch but now with 6.1 and deleting instead of overwitting the new version of Reinx everything works perfectly.

Thanks a lot, you can close the issue :)

closed 2019-02-25 by PaS-NaS