Reisyukaku/ReiNX #101: what's mean the partial 7.0.0 support?

TheOneEyedGrimReaper, 2019-02-10:
my little brat updated my switch to 7.0.0.
i used sx os.
i tried the lastest sx os and your reinx, but in the sx os black screen after all, with your reinx, the console is freezed on the reinx logo.

how much we can wait for the full 7.0.0 support?
what we can expect? 1-2 week or 1-2 month?

Reisyukaku, 2019-02-10:
Hard to say time wise. We are still missing some of the keys we need.

TheOneEyedGrimReaper, 2019-02-11:
> Hard to say time wise. We are still missing some of the keys we need.

thanks for your reply.
is this nice.

i have a question.
can i use another console emunand?
and this tutor is working to downgrade my switch or fake? [https://guide.sdsetup.com/#/manualchoidowngrade](url)

videogiocatore, 2019-02-14:
Where we can follow the new updates for the future support of 7.0 firmware? :D 

Main page of https://reinx.guide/, twitter, patreon, here?

videogiocatore, 2019-02-27:
> > Hard to say time wise. We are still missing some of the keys we need.
> 
> thanks for your reply.
> is this nice.
> 
> i have a question.
> can i use another console emunand?
> and this tutor is working to downgrade my switch or fake? [https://guide.sdsetup.com/#/manualchoidowngrade](url)

I just used atmosphere 0.8.4 (partial support 7.0.1) + tinfoil 1 and with some work i did it even installed 3-4 nsp :D (first time never modify my switch before and i had it at 7.0.1) i used this ready to use folder.

https://www.reboot.ms/forum/threads/modifica-funzionante-atmosphere-7-0-1-finalmente.11265/

I tried with original atmosphere 0.8.4 + tinfoil but nsp had problem installing this ready to use folder to put on sd has already patches to atmosphere + keys.txt and everything works perfectly (installed pokémon, undertale, SSB ultimate and FF IX)

Reisyukaku, 2019-03-19:
Full support added

closed 2019-03-19 by Reisyukaku