Reisyukaku/ReiNX #109: Updated ReiNX from 2.0, no splash screen

capSAR273, 2019-03-17:
So I moved the sept and ReiNX folders over to the SD Card, removed the old copy, and injected using TegraRCM, but no splash screen (though it says it injected properly). Should I stay on 2.0 if I haven't updated to firmware 7.0 yet?

Reisyukaku, 2019-03-19:
if you're not on 7.0, it doesnt even use sept.. everything should still boot fine though. What fw?

capSAR273, 2019-03-19:
Should be 6.2 (I think), I went back to 2.0 ReiNX for now and I'll check the fw for sure later today.

nection, 2019-03-20:
When loading the payload it remains in black screen. And I do not have the * .bin in the SD. Any idea how I can solve? Atmosphere + Hekate if that starts me well.

Reisyukaku, 2019-03-20:
@nection delete reinx folder and place the new on on the sd.. thats it.

nection, 2019-03-20:
> @nection delete reinx folder and place the new on on the sd.. thats it.

I've explained myself wrong, my English is horrible and I have not read the advertised, I come from a clean inatalacion of the latest version of Reinx and I'm in 7.0.1, the screen goes black when I inject the paylod, I read Many people have the same problem.

Reisyukaku, 2019-03-20:
Dont put the ReiNX.bin on the sd. Only the folder. My testers can't replicate this issue.

nection, 2019-03-20:
> Dont put the ReiNX.bin on the sd. Only the folder. My testers can't replicate this issue.

Yes, I knew it, it's the first thing I tried but it still does not start.

TrevTV, 2019-03-20:
Hi! I had this same issue using the ReiNX.bin file. I replaced my Atmosphere sept files with ReiNX's. The ReiNX bin wouldn't work but when I tried with Atmosphere's fusee-primary.bin it booted into ReiNX working just fine. I was able to install nsps (I tried the same nsp on Atmosphere aswell, didn't work). So I suggest trying fusee-primary to boot instead of ReiNX.bin.

flxdhc, 2019-03-27:
I encountered the same issue, I went back to 2.0 (using 6.2). My SD is in Exfat format, and I am using a macbook. Both web fusée and terminal fusée gelée do not proceed after the step "triggering vulnerability". The 2.0 package is working very well. Do you have any advice for this situation?

patchper, 2019-03-29:
I have the same issue install a fresh ReiNX 2.1 on firmware 5.1.0, but 2.0 worked well.
Injection was sucessful but no splash screen was shown.

petabyteboy, 2019-03-30:
I am having the same problem. Using ReiNX 2.1 without the ReiNX.bin on the SD card, fusee launcher seems to run successfully, but the screen stays black, no splash screen, nothing. My console i running 7.0.0.

Reisyukaku, 2019-03-31:
fixed as of 22830d5f9bd9099ede7ec035d1c848cb4d822ad8

closed 2019-03-31 by Reisyukaku