Reisyukaku/ReiNX #114: Splash screen shows but black screen after that

capSAR273, 2019-03-31:
So, I saw that #109 was fixed in the latest hotfix, I loaded up the new version but this time, I get a black screen after the splash. Not sure if I can provide any logs, TegraRCM just says:

`Uploading payload (mezzo size: 92, user size: 76598, total size: 142814, total padded size: 143360)...
Smashing the stack!
Smashed the stack with a 0x7000 byte SETUP request!
Payload successfully injected`

I'm on 6.2.0 FW, all that was changed was the ReiNX version from 2.0 to 2.1.1

Reisyukaku, 2019-04-01:
try the latest release as of basically right now.. you might have a jank dl

flxdhc, 2019-04-01:
I had issues with the 2.1 release (#109). I updated to the 2.1.1 release. I have the splash screen but after it, it boots Horizon OS instead not the CFW. I am still on 6.2. Should I update to 7.0.1?

RetroGamer74, 2019-04-01:
Hotfix 2.1.1 is wrong. It reboots after SEPT. In my case back to RCM ( because of AutoRCM ). Fix required.

Reisyukaku, 2019-04-01:
@EliseZeroTwo

Reisyukaku, 2019-04-01:
@flxdhc if you're booting into HOS, thats cfw. Theres no mechanic for booting into unmodified firmware.

Reisyukaku, 2019-04-01:
@RetroGamer74 You're required to have ReiNX.bin on the SD card

RetroGamer74, 2019-04-01:
I think he means, because after SEPT, ReiNX reboots, then OFW starts its loading.

RetroGamer74, 2019-04-01:
Where?

Reisyukaku, 2019-04-01:
If you've copied everything in the folder to the SD (preferably after deleting existing reinx files), everything should work. A lot of people either dont drop everything from the release on to SD, or some people dont clean out old files and have weird issues because old and new are running together.

RetroGamer74, 2019-04-01:
Ok. I'm gonna try again.

RetroGamer74, 2019-04-01:
It works!!! The issue was ReiNX.bin. Why now is mandatory on the root of SD?

Reisyukaku, 2019-04-01:
Because sept is very fussy and juggling around the payload was causing issues, so the solution was to just
reinx -> sept -> custom payload -> reinx.bin off sd

RetroGamer74, 2019-04-01:
I keep updated a boot manager which is RetroReloaded. I don't want to do spam, but now in my case it does:
Custom Payload (RR) -> /RR/payload/ReiNX.bin -> Sept ->custom payload -> /ReiNX.bin

So I've got two times your ReiNX.bin. Further than this, everything works ok!!.

capSAR273, 2019-04-01:
So if I'm on a FW under 7.0, no sept folder required? I moved the .bin to the SD root this time to test, but am still getting a black screen. I wiped out the old ReiNX folder also, replaced with new version, no luck.

Reisyukaku, 2019-04-01:
you probably forgot to include the hbl.nsp.. its on our guide. I've never repacked it with the cfw. Also i'm using 2.1.1 fine on 6.2 without sept.

flxdhc, 2019-04-02:
> @flxdhc if you're booting into HOS, thats cfw. Theres no mechanic for booting into unmodified firmware.

@Reisyukaku thanks, all good for me, I had not noticed that hbl.nsp was needed...

closed 2019-04-02 by Reisyukaku