Reisyukaku/ReiNX #116: Homebrew menu not launching

bonezinger, 2019-04-08:
Running on FW 7.0, ReiNX 2.1 (also tested on 2.1.1)
I can't open the homebrew menu. I have the hbmenu.nro in the root of the sd card, but whenever I open the album, it just opens normally. I've tried holding R, L, ZL and ZR, but nothing. I've checked the config file, _/config/hbl-config.ini_, it's still the default version.

I got the /switch folder with other .nro files, so I don't think the problem is folder structure.

Reisyukaku, 2019-04-09:
either use the hbl.nsp in /ReiNX folder or if you want to use nro, use the legacy method posted in our guide 
https://reinx.guide/guide/download-and-install
under optional downloads

bonezinger, 2019-04-10:
Turns out I was lacking the hbl.nsp in the proper place. Thanks for the help!

closed 2019-04-13 by Reisyukaku