Reisyukaku/ReiNX #117: 100% games launching on fw 7.0.0 failing

tantoinet, 2019-04-09:
All of the successfully installed (from and to sdcard) nsp games did not start, with the generic error message: "Could not start the software. Please try again from the HOME menu".
Didn't try the USB installer.
Details:
Firmware: 7.0.0
ReiNX: 2.1.1 (/patches/es_patch.rxp 2d21aa7238bade50dfb7113cdb7d96be. Directory tree validated with discord support channel, no missing file)
NSP installers: Tinfoil 1.5.5 (from tinfoil.io) or Goldleaf 0.5
Micro SD Card: fresh 128gb SanDisk sdcard ordered on Amazon

Fix: upgrading to fw 7.0.1 fixed the issue.

Reisyukaku, 2019-04-09:
7.0.0 and 7.0.1 are virtually identical. My es patches work using pattern searches, so i dont see why there would be a discrepancy there, but i can look into it.

Nettozx, 2019-04-23:
I had this problem and I added the patches in this thread and it worked:
https://gbatemp.net/threads/i-heard-that-you-guys-need-some-sweet-patches-for-atmosphere.521164/

closed 2019-04-23 by Reisyukaku