Reisyukaku/ReiNX #126: Sleepmode doesnt work

Peronia, 2019-04-25:
With reinx 2.2 and 8.0.1 my sleepmode is gone.
In discord im not alone.
When I press power short my switch went black. When I press it again, my switch stay black. Only a hardreset can wake it up.

sugeinmy, 2019-04-25:
the same

jit06, 2019-04-25:
The same for me. I was in 6.2 before upgrading...

lavaJockey, 2019-04-25:
This is fixed in the new commit. If you like using builds it will fix the problem.

Peronia, 2019-04-25:
I would test it but I cant compile it

jit06, 2019-04-25:
I just built it with docker, replaced the files on the SDcard and used the new Reinx.bin to boot. But I still get the same symptoms : if I switch push the power button while on, the console never wake up.

Reisyukaku, 2019-04-26:
fixed and released a hotpatch for it

closed 2019-04-26 by Reisyukaku