Reisyukaku/ReiNX #129: game doesn't open, requires internet connection to run.

MUGIWARA93, 2019-04-27:
install two games and both asked to disable the airplane mode and then asked to connect to the internet to open it, installed well and so far is stable, the only thing is to connect to internet with all games.

MUGIWARA93, 2019-04-27:
[image]
[image]
[image]

I clarify that I already have the dns changed and I am in 8.0.1 with ReiNX 2.2.1

Reisyukaku, 2019-04-28:
Fixed. Download the new rxp and replace the one in the patches folder.

closed 2019-04-28 by Reisyukaku