Reisyukaku/ReiNX #139: Blackscreen after ReiNX logo

fenomeno0chris, 2019-06-27:
Hi, after following the stpes on the offical ReiNX website Im stuck on a black screen. The ReiNX logo just pops out a few seconds and than its black and it doesnt boot to the homescreen.

Is there something wrong with 2.4, cause Im using a freshly formatted microSD card. Ive also tried both exFat and Fat32.

Switch Info: 
Version 7.0.1
Tried with hekate and reinx payload

NoyTheBiche, 2019-07-02:
Try [this](https://cdn.discordapp.com/attachments/548845841306091520/591663817835151380/ReiNX_2.4_Fixed.zip) (from ReiNX's discord)

Rotule666, 2019-07-02:
> Try [this](https://cdn.discordapp.com/attachments/548845841306091520/591663817835151380/ReiNX_2.4_Fixed.zip) (from ReiNX's discord)

It didn't work for me, had the same issue and Using Version 7.0.1 as well

sli, 2019-07-11:
Getting this on 7.0.0 after updating to 2.4 and going back to 2.3 didn't fix it, either.

fotiDim, 2019-07-21:
> Try [this](https://cdn.discordapp.com/attachments/548845841306091520/591663817835151380/ReiNX_2.4_Fixed.zip) (from ReiNX's discord)

Same problem here. The suggested files didn't help. Tried with both FAT32 and exFAT.

KnightMars1, 2019-08-01:
Confirmed 2.4 Renix is incompatible with 7.0.1 fw?

lookye, 2019-09-22:
ReiNX 2.4 works correctly with FW7.0.1 is you set up your files fine.
If you dont know how to set them up correctly, visit the guide (www.reinx.guide).

paishin, 2019-09-25:
I had the same issue. Turns out it was caused by me incorrectly upgrating to the fat32 version while my sdcard eas formatted to exfat. I formatted the card to fat32 and reloaded reinx files and everything booted just fine. My only issue is that now switch drains battery on standby something that did not happen on exfat.