Reisyukaku/ReiNX #156: Sigpatches

FellFromTheSky, 2020-05-08:
If I'm right :
KOSMOS patches (dead) supported nocmac patches, fs_patches, ACID patches es_patches
NEUTOS (dead) supported fs_patches, ACID patches and es_patches
HarukoNX PKG2 patches still support fs_patches, ACID patches and es_patches 

So which patches are suppoted in ReiNX?
If you wantto support other patches, I've still a backup NEUTOS and Kosmos patches on my computer, and HarukoNX PKG2 patches are still availableon GitHub.

D3fau4, 2020-05-17:
you dont need other patches because they are already included in the system but in any case, the exefs patches goes in: `/ReiNX/patches/`

FellFromTheSky, 2020-05-17:
Maybe yes, because you can only load Rei

closed 2020-05-31 by FellFromTheSky