Reisyukaku/ReiNX #157: Report : won‘t work with idn_mitm

Painkiller12345, 2020-05-15:
Hello,
I am a newer of ReiNX, 
I followed the guide from the page here:https://reinx.guide/  and have settled everything succesefully except the Sysmodule:ldn_mitm 

guide Step 3 - Post-Installation mentioned:
"When installing sysmodules please keep in mind most of them come inside a /atmosphere/ folder by default, just copy the contents folder from this to the /ReiNX/ folder on the root of your sd card."

so i download ldn_mitm from:https://github.com/spacemeowx2/ldn_mitm/releases
there is a folder named 4200000000000010 under the atmospher/contents folder which i unzip from the spacemeowx2's latest release ldn_mitm_v1.3.4.zip
then i put the 4200000000000010 folder into the /ReiNX/ folder/contents folder of my sd card. 

after that i reboot the system ,when i try to get into the HBmenu with the Photo album button,
my switch turn out black screen and  reboot to reinx automatically ，tap the Photo album button again it still reboot the system automatically,and the idn-mitm sysmodule would not show in the tesla menu
after i delete the folder ReiNX/ folder/contents/4200000000000010 it works well
my switch was baned by nintendo and i want to play on the LAN Play server 
long time ago i use to hack my switch with another cfw and the idnmitm works fine,i can find the background service on in the tesla menu
is there some bugs unfixed between ReiNX and idn-mitm or am i doing someting wrong?
please help me out

D3fau4, 2020-05-17:
I think your SD is corrupt

Painkiller12345, 2020-05-19:
> 
> 
> I think your SD is corrupt

what am i suppose to do then?

Painkiller12345, 2020-05-22:
> 
> 
> I think your SD is corrupt

YOUARE RIGHT ,ITS MY OWN FAULT,I HAVE SOLVE THE PROBLEM BY REINSTALL REINX,NOW I CAN USE IDN-MITM,THANK YOU VERY MUCH.

closed 2020-05-27 by Reisyukaku