Reisyukaku/ReiNX #313: KIP hash not found in list! (10.2.0)

erulogoss, 2020-09-19:
error when updated to 10.2.0
"KIP hash not found in list!"

How to repair this problem ?

iball, 2020-09-19:
Seeing the same issue as well.

NaturalHarmonia, 2020-09-22:
I accidentally updated and I'm getting this as well.

gradius3, 2020-10-02:
Same problem for me. Solution needed. And strange thing it happened with airplane mode on!?!

Adran-Marit, 2020-10-06:
Downgrade back to 10.1.0 until such time as the ReiNX bootloader is updated. 

This can be done by temporarily switching to Atmosphere and using daybreak, choidujourNX to install 10.1.0

gradius3, 2020-10-07:
Thanks a lot for this workaround solution. Hopefully it will not take too long this time until a little reinx update arrives!

MARIOWARE, 2020-10-13:
> error after update to 10.2.0
> "KIP hash not found in list!"
> 
> How to repair this problem ?

I found out that if the error pops up all I have to do is press the volume up button. And once  a message pops up press the volume up again. And now it works perfectly fine. I don't understand the problem or why it works when I press the volume up button but so far no problem.

gradius3, 2020-10-13:
> I found out that if the error pops up all I have to do is press the volume up button. And once a message pops up press the volume up again. And now it works perfectly fine. I don't understand the problem or why it works when I press the volume up button but so far no problem.

That's mostly correct. What is not running anymore are "repacked" nsps.  With the program reNXpack you had the possibility to patch games, so they were usable with reinx when there wasn't a firmware update available (End of 2019). With a repack game there are multiple random nca files written inside Nintendo/contents/registered and its not possible to tell which files belong to which game. And due to these files installed in the system one can't just delete the game and overwrite them with untouched nsps as it gives you an error message. 

Another problem is AM2R nsp which isn't playing anymore.

MARIOWARE, 2020-10-13:
> That's mostly correct. What is not running anymore are "repacked" nsps. With the program reNXpack you had the possibility to patch games, so they were usable with reinx when there wasn't a firmware [...]

Sorry I don't really understand. See you saying that I cant delete or play some games ?

gradius3, 2020-10-15:
> 
> 
> > > I found out that if the error pops up all I have to do is press the volume up button. And once a message pops up press the volume up again. And now it works perfectly fine. I don't understand the problem or why it works when I press the volume up button but so far no problem.
> > 
> > 
> > That's mostly correct. What is not running anymore are "repacked" nsps. With the program reNXpack you had the possibility to patch games, so they were usable with reinx when there wasn't a firmware update available (End of 2019). With a repack game there are multiple random nca files written inside Nintendo/contents/registered and its not possible to tell which files belong to which game. And due to these files installed in the system one can't just delete the game and overwrite them with untouched nsps as it gives you an error message.
> > Another problem is AM2R nsp which isn't playing anymore.
> 
> Sorry I don't really understand. Are you saying that I cant delete or play some games ?

It seems to work all fine as long as you never installed modified NSPs. If you have previously installed modified games such as the mentioned "repack" versions or i.e. AM2R you won't be able to play these anymore.

MARIOWARE, 2020-10-15:
> It seems to work all fine as long as you never installed modified NSPs. If you have previously installed modified games such as the mentioned "repack" versions or i.e. AM2R you won't be able to play [...]

So I won't be able to play for example: fan made switch games ? But there's no other problem except for fan made switch games ? Like If the new botw came out I would be able to play it right ? Sorry for asking so much I'm new to switch hacking.